MXCHIP-EMW/WICED-for-EMW #1: Bootloader patch broken

emilf, 2015-08-15:
i have cloned WICED-for-EMW and got an error message while trying to patch: https://gist.github.com/rhaamo/d735cb883a8d07119b53
about the boot loader patch

rhaamo, 2015-08-15:
I've tried to patch against 3.1.0, 3.1.1 and 3.1.2 and can't get the patch applied, and haven't found infos on where to put the snippet.
Can you try to do the next patch using "diff -Naur old new" as options to export the patch ?
(it should be more-readable that the default "diff" output)
Thanks.

emilf, 2015-08-15:
The patch is supposed to be for 3.1.2 and always worked fine for me. Thanks for the pointer on the diff command line.

emilf, 2015-08-15:
I just looked at this on my machine and it looks like the bootloader patch worked just fine on my machine.

This is from my bootloader.c on 3.3.1:
```
int main( void )
{
    const load_details_t* load_details;
    uint32_t              entry_point;
    boot_detail_t         boot;
    
    // Patched for the EMW3162. Hold the button to halt booting.
    // This enables flashing, even if the application uses the JTAG
    // for something else.
    if ((!wiced_gpio_input_get( WICED_BUTTON1 )) && (!wiced_gpio_input_get( WICED_BUTTON2 ))) {
        while(1);
    }
    // End of patch.
#ifdef PLATFORM_HAS_OTA
    NoOS_setup_timing( );
```

This is my patch binary info:
```
$ patch --version
GNU patch 2.7.5
Copyright (C) 2003, 2009-2012 Free Software Foundation, Inc.
Copyright (C) 1988 Larry Wall

License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>.
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.

Written by Larry Wall and Paul Eggert
```

rhaamo, 2015-08-16:
The #include seems to be added properly but the if no.
I've recreated a patch (below) and for me it applies properly.

```patch
--- WICED-SDK-3.3.1/apps/waf/bootloader/bootloader.c	2015-06-29 12:33:36.000000000 +0200
+++ WICED-SDK-3.3.1/apps/waf/bootloader/bootloader.c.new	2015-08-16 09:32:17.000000000 +0200
@@ -18,6 +18,7 @@
 #include "platform_resource.h"
 #include "waf_platform.h"
 #include "wwd_rtos.h"
+#include "wiced_platform.h"

 /******************************************************
  *                      Macros
@@ -59,6 +60,14 @@
     uint32_t              entry_point;
     boot_detail_t         boot;

+    // Patched for the EMW3162. Hold the button to halt booting.
+    // This enables flashing, even if the application uses the JTAG
+    // for something else.
+    if ((!wiced_gpio_input_get( WICED_BUTTON1 )) && (!wiced_gpio_input_get( WICED_BUTTON2 ))) {
+        while(1);
+    }
+    // End of patch.
+
 #ifdef PLATFORM_HAS_OTA
     NoOS_setup_timing( );
```

emilf, 2015-08-28:
I've disabled the bootloader patch for now... especially as the EMW3165 only has one user button. I've yet to check it in, but it's coming :p
I've also decided that it shouldn't be in the bootloader anyway, but rather in the platform peripheral init... then we keep WICED itself platform agnostic.

closed 2015-09-25 by emilf

emilf, 2015-09-25:
Heh, the bootloader patch is disabled with an #ifdef, so the commit really does nothing.
But it's an option for those who need it.

commit 004072c5 (2015-09-25): Fixes #1 for EMW3162. Not doing the same for EMW3165, as the devboard only has one button.