MXCHIP-EMW/WICED-for-EMW #3: ./make EMW3165 fails

rogerclarkmelbourne, 2015-08-16:
Hi,
I'm trying to follow your build instructions and I downloaded WICED and used the script to extract and patch it.

You instructions didnt mention you then need to go into the newly extracted WICED folder (but never mind)

Anyway, I tried runing 
./make EMW3165 

but I get this error

./tools/makefiles/wiced_config.mk:282: *** No application specified. Options are: README.txt demo snip test waf wwd.  Stop.
make: *** No rule to make target `build/EMW3165/config.mk', needed by `main_app'.  Stop.

I get the same message for ./make EMW3162

So I tried

./make EMW3165 #>-LwIP-FreeRTOS-snip.apsta download run

But I still get the same error

I can see that there is a platform/EMW3165 folder
but running make in there doesnt seem to work either

Can you clarify the build instructions?

emilf, 2015-08-16:
On Sun, Aug 16, 2015 at 8:30 AM, Roger Clark <notifications@github.com>
wrote:

> ./make EMW3165 #>-LwIP-FreeRTOS-snip.apsta download run

Try:
./make EMW3165-LwIP-FreeRTOS-snip.apsta download run

It should build, but "download run" will fail unless you magically managed
to connect an FT2232H to the non-existent standard JTAG port.
Thanks for your comments, I'll clarify the instructions. I'm also working
on actually testing this on the new EMW3165. The instructions are mostly
based on the older EMW3162 experience and I've only been able to start
playing with the EMW3165 recently.
The EMW3165 requires J-Link to program, as the default option doesn't work
for the new board. I'm actually wiring up a busblaster right now to see if
I can use that to properly program the EMW3165. It was a problem with the
older WICED version I uved before, as the custom OpenOCD that came with it
only supported the old FT2232 driver, rather than the newer FTDI driver.
You should be able to use St-link v2 to flash, using a slightly modified
flash.sh script, as the build output location has changed.

rogerclarkmelbourne, 2015-08-16:
I have a ST-Link and also a JTag. But the Jtag is a Chinese clone, and I've never had much success getting it to work.

I did get it to compile, except I then didn't have OpenOCD installed, so I got an error message.
So I've now installed OpenOCD, but had to stop for today (as its 9pm here)

At the moment I know nothing about WICED to I will need to read the overview and some of the documentation before I can proceed any further.

Re: STLink
I can flash the EMW3165 using STLink in Windows using STM's own ST-Link utility, but on Linux I'll probably have to use Texane/Stlink - or perhaps OpenOCD supports ST-Link

emilf, 2015-08-16:
There is a script in the repo called flash.sh... it should work to flash
with the stlink using openocd. Also, there are openocd bwinaries with wiced
under tools/OpenOCD/Win32.

I've been able to flash with stlink using this script, though when I did,
then there were other issues in the build that made what I flashed not
work, so I've yet to verify that the stlink flash is actually doing it
correctly.

I'm working on all of the jtag/swd support right now, aiming for Bus
blaster (at least v4, w. kt-link firmware), st-link v2, jlink (chinese
clone, hw version matters and clones under a certain version actually short
something out if you try to put them in SWD mode) and perhaps even FT2232H,
using an ugly hack to have SWD support without buffers.
Perhaps even more jtag adapters, if I have 'em or can emulate 'em with the
bus blaster.

St-link v2 was a headache with the WICED 3.2.1 due to how the included
version of openocd was configured. I think the openocd version in WICED
3.3.1 has a newer type of stlink drivers that allow using stlink without
having separate target definitions and if that's working, I might make
stlinkv2 with WICED.
If that does work, I think I'll put the St-link v2 as the preferred
jtag/swd adapter for the project and put the most focus on supporting that.

rogerclarkmelbourne, 2015-08-16:
Thanks for the information.

STLink would be my preferred programmer, as I have used them before and they are small and cheap.

Another option however is the Black Magic Proble, which is open source and can be flash onto a cheap $5 STM32F103C board. BMP works well with OpenOCD.

I have tried using my Jlink clone with other STM32 boards, but although it appears to upload the code, I could not get it to actually run .

You mentioned Win32 versions...

I would prefer a Win32 build, so i guess I will need to work out what your "extract and patch" script does, and replicate it on Windows.
I have make.exe on windows, but i guess it is also going to be more of a challenge to get it to build on Windows.

emilf, 2015-08-17:
If you run the extract-and-patch script under Cygwin, it probably works
just fine.
I'm actually having the same issue with getting code to run properly and am
diagnosing that.

rogerclarkmelbourne, 2015-08-18:
I installed the WICED IDE for Windows, which appears to be a branded version of Eclipse

And I've taken a look at your "extract and patch" script, which generally seems to just copy files.
The only patch I can see is the bootloader.c patch, is this correct

The Windows WICED SDK its self looks very similar to the 7Z file, I suspect its identical, except perhaps for the .project file

Strangely, Ecluipse is showing an error on wiced_config.mk even before I attempt to patch the SDK by hand, (ie do what your script does)
But at the moment I'm not sure why Eclipse is showing a cross (x) against that file.

Perhaps its not important. ;-)

emilf, 2015-08-28:
Sorry about not answering earlier... there are some issues with how Eclipse analyses the code, so in many cases you get errors or warnings about unresolved symbols and stuff like that in the IDE, but the build works fine.
Nb. there is preliminary stlinkv2 support in the repo now, though the patcher script doesn't automatically update it, as it's not properly tested with all configurations.
It works to both flash and debug. WICED apps don't actually work yet, but the bootloader does run and I think I might have found the problem at long last, so hopefully we can start claiming proper support for EMW3165 soonish.

About the WICED IDE having an older version of WICED, I recall seeing a blog post on the Broadcom WICED site about how to go about installing a newer version of the SDK in the IDE, so you might look there.

rogerclarkmelbourne, 2015-08-28:
OK.
Thanks.
I'll try to take a look during the weekend.

My breakout boards finally arrived from Dirty Cheap PCB's this afternoon, so I may attempt to hand solder one of my EMW3165's onto a breakout - though I'm not sure how feasible that's going to be, or whether I need to use my reflow kit.

emilf, 2015-08-28:
FYI.... almost everything should work now.

I hand soldered mine, but I recommend supergluing the module to the board first, making sure its very well aligned. I've tried starting soldering the corners to tack it down, but it always seemed to get misaligned somehow and it's just too easy to short between those tiny pads.
Soldering the dirtypcb breakouts was a lot easier than the MXCHIP devboard... that really is a bitch to get right without lifting pads or shorting stuff.

To build and flash via stlink-v2 do this:
./make EMW3165-LwIP-FreeRTOS-snip.apsta download run JTAG=stlink-v2

If you make any changes to improve windows support or documenatation, please send in PRs.

closed 2015-08-28 by emilf