MXCHIP-EMW/WICED-for-EMW #5: Verify various JTAG/SWD adapters on both EMW3165 and EMW3162 with OTA_FR support

emilf, 2015-08-28:
JLink, Stlink-v2, ft2232h on EMW3162 only, busblaster v4 w/ ktlink and more if I have 'em.

emilf, 2015-09-25:
For now, the EMW3162 green development board built-in FT2232H and stlink-v2 are the two best supported JTAG/SWD adapters.

J-link and stlink-v2 work on EMW3162 and EMW3165 on Linux.
Busblaster v4 w/ ktlink buffer does not have the same pinout as the jtag connector on the devboard IIRC, so that destroyed one EMW3165 module for me. It can probably work, but I've not bothered.

FT2232H cannot* work with EMW3165, as it requires SWD, but does work with EMW3162 and is on the green EMW3162 devboard, with a custom configuration in EEPROM. Details can be found on the WICED blog.

I might check out if I can make a Cypress FX2LP devboard work. I know it's used as JTAG for a Digilent Nexys 3 FPGA development board and I have a few laying around. If anyone is interested, poke me  and I'll test it out.

*I've read somewhere that adding a resistor between some pins and hoping for the best might work, for unbuffered SWD, but that is a rather hacky stuff and I'm not even going to attempt it.

leppie, 2015-09-28:
ST Link v2.1 (as found on Nucleo boards) also works. You just need to add the interface cfg from openocd. Only difference is the USB PID.

emilf, 2015-09-28:
Yeah. Forgot to add it here, but I've already added the openocd config file
for it to the repo. Use JTAG=stlink-v2-1 on the make command line to use
it.