MXCHIP-EMW/WICED-for-EMW #6: Investigate SPI flash OTA support on EMW3165

emilf, 2015-08-28:
It builds and tries to flash, but something is causing hardfaults on download_apps.

emilf, 2015-09-25:
download_apps works now, uploading an update via ota_fr works.

However, the new app never boots for some reason.

So making progress slowly... it's hard to debug over hardware resets and not knowing what image is running.

emilf, 2015-09-25:
Note to self... changes to wicked_apps.mk(WICED bug fix) and EMW3165.mk (for saving the wifi driver binary blob in app slot rather than internal resources, for multiapp support)

leppie, 2015-09-28:
I was testing this over the weekend before my module just gave up.

SPI flash speed of 8000000 works most of the time. But also got issue running the apps after upload via web UI.

BOB62, 2015-11-10:
Get-it after a few hours of debug, 
you must adjust the end of sector in the waf_platform.c in the directory Stm32F4xx/WAF 

#define PLATFORM_APP_START_SECTOR      ( FLASH_Sector_3  )
#define PLATFORM_APP_END_SECTOR        ( FLASH_Sector_7 )  instead of 13

emilf, 2015-11-13:
OMG! Testing now... my god I've spent a lot of time on this shit....
If this works I'll build a statue in your honor

emilf, 2015-11-13:
So far so good... patch coming up :)

commit 7af21344 (2015-11-13): Afaik fixes #6, but needs more testing. At least flashing snip.ota_fr and then using it to ota flash snip.apsta works, but I've not tested actually entering factory reset mode, just because of the way my hardware is set up right now.
closed 2015-11-13 by emilf