MXCHIP-EMW/WICED-for-EMW #8: problem with Segger J-Link

ammaree, 2015-09-25:


emilf, 2015-09-25:
Ok.

You might have to override the default jlink drivers with libusb ones.
If that doesn't work, please try the following:

Can you give me the output of running `start /B ./tools/OpenOCD/Win32/openocd-all-brcm-libftdi.exe -f ./tools/OpenOCD/jlink.cfg`
and then install OpenOCD separately and run `openocd -f interface/jlink.cfg`

emilf, 2015-09-25:
WICED forums issue: https://community.broadcom.com/message/1133#1133
This points to using the libusb driver and gives directions.

emilf, 2015-09-25:
Direct link the the post with the instructions: https://community.broadcom.com/message/7920#7920

emilf, 2015-10-12:
Platform specific, non-project specific issue.

closed 2015-10-12 by emilf