MXCHIP-EMW/WICED-for-EMW #9: Not all SPI interfaces are accessible on EMW3165 (WICED bug really)

emilf, 2015-09-25:
From Gitter chat:
@paulreimer

paulreimer
I have a question about using the SPI4 + SPI5 interfaces on the EMW3165: I see they are present on the STM32F411xE, and pins are available on the 3165. However, support for these is missing from the STM library code included (it only supports up to SPI3). Also it looks like the STM code is pre-STM32Cube+HAL. What is the best way for me to add support for these interfaces? Back-fill the WICED STM code with the IRQs, logic, enums, etc in the old style? Somehow port the code to STM32Cube and replace with those sources?
If anyone knows if a future WICED version supports SPI4 and SPI5, maybe I can selectively backport to this 3.3.1-based project? Hopefully I am missing something, but it looks like SPI2 and SPI3 don't have the MISO and/or MOSI pins broken out on the EMW3165

emilf Sep 22 19:56
tbh, I don't really know much about how the standard STM library code works and I've only used STM32Cube to help me validate clock configurations... where (what files) should I look at to start investigating?
You are referring to the SPI support in the STM libraries, right? Not WICED or the EMW3165 platform configuration, that only has one SPI bus defined in it...?

paulreimer Sep 23 04:35
Yup, I'm referring to the STM library code. For example: WICED/platform/MCU/STM32F4xx/peripherals/libraries/src/stm32f4xx_spi.c
Try searching for SPI3 in the WICED root directory, and see how STM supports it.
I already added the SPI2~SPI5 pins to the platforms/EMW3165/platform.c file (this approach worked well for adding SPI2 + SPI3 on the EMW3162: it worked!)

emilf, 2015-09-25:
@paulreimer
This is on the WICED forums:
https://community.broadcom.com/message/12351#12351

If you make those changes, a pull request would be appreciated.
Otherwise, I'll get around to it when I have time...

paulreimer, 2015-09-25:
Hi Emil,

Thanks that helps a lot! I had made the changes last night for the STM portion (memory addresses, IRQs, register masks, etc), but it didn't boot. I was in the middle of bisecting which chamge broke it, and I was getting close but had to sleep.

I'll take another look tonight and add any info from that ticket.

Cheers and thanks,
Paul

emilf, 2015-09-25:
Happy to help :) It's easy to forget looking at the WICED forums for any issues, as it's not indexed by Google :p

emilf, 2015-10-22:
#13 deals with this. Not merged yet, still waiting on feedback.

emilf, 2015-11-03:
I've merged #13. I'll add it to the patch script as well. RX DMA seems not to work, but TX does and both work without DMA

mhowarddune, 2015-12-14:
Hi,

Has anyone worked with the RX DMA issue?  I would like to use this and have time to look into the problem.

Regards, Mike

paulreimer, 2015-12-14:
Hi @mhowarddune @emilf ,

I can confirm that the STM32 stdperiph patch works correctly for SPI RX and TX DMA. I'm using it successfully for both RX on SPI4 and TX on SPI5 (and I also verified TX on SPI4).

I'm using direct StdPeriph code instead of the WICED functions since my needs are a bit different (16-bit, circular, RX-only, w/ISR). This works for me, so I'm thinking any problems are within the WICED portion (and quite possibly more fundamentally flawed than a typo). Unfortunately I cannot provide code samples from my project, but the STM32 forums are pretty good for obtaining snippets.

Cheers,

Paul

Cheers,Paul

mhowarddune, 2015-12-14:
Thanks, that information is helpful.  I will take a look.

Mike

emilf, 2015-12-15:
I think this is fixed in WICED 3.4.0, though have neither properly ported the project to it nor verified that it works. I just vaguely recall seeing something like this when I started looking at what had changed, but it's been a few weeks so I might be mistaken.
I'm on a bit of a holiday hiatus from working on the project at the moment, so it'll be, probably, at least a couple of weeks until I get going on this again. (and hopefully my newfound Fallout: New Vegas enthusiasm has died down a bit by then :p )

mhowarddune, 2015-12-19:
It looks the the DMA channel is set wrong for RX DMA SPI4 in platform.c
(AN4031 pg 9 says it should be Channel 5, not 4)  and I do not see a
definition for SPI6, I will check these in the 3.4 source.

Mike Howard

paulreimer, 2015-12-19:
Interesting, that document (AN4031) has a different DMA table from the one I used:

Reference Manual (RM0383), page 167
http://www.st.com/web/en/resource/technical/document/reference_manual/DM00119316.pdf

The reference manual is specifically for "STM32F411xC/E" MCUs (and plus, I am getting working transfers with DMA2_Stream4/DMA_Channel_4).

Also if you check page 37/38 in that document (register boundary addresses table), I'm not sure there is an SPI6 expected on this MCU.

As a second question though, WICED 3.4.0 sounds interesting; but I can't find it on the WICED forums, I'd like read the Release Notes at least!

Novatec1, 2016-06-09:
patch for SPI5 on emw3165 is confusing. Official  mxchip document says GPIO_29 as   SPI5_MOSI, while the patch is using GPIO_31