MXCHIP-EMW/WICED-for-EMW #12: Flashing not possible on Mac OS X

eliabieri, 2015-10-12:
I get this Error when I want to flash the EMW3165 on Mac OS X Yosemite

    Downloading Bootloader ...
**** OpenOCD failed - ensure you have installed the driver from the drivers directory, and that the     debugger is not running **** In Linux this may be due to USB access permissions. In a virtual machine it   may be due to USB passthrough settings. Check in the task list that another OpenOCD process is not running. Check that you have the correct target and JTAG device plugged in. ****
Downloading DCT ...
**** OpenOCD failed - ensure you have installed the driver from the drivers directory, and that the debugger is not running **** In Linux this may be due to USB access permissions. In a virtual machine it may be due to USB passthrough settings. Check in the task list that another OpenOCD process is not running. Check that you have the correct target and JTAG device plugged in. ****
Downloading Application ...
**** OpenOCD failed - ensure you have installed the driver from the drivers directory, and that the debugger is not running **** In Linux this may be due to USB access permissions. In a virtual machine it may be due to USB passthrough settings. Check in the task list that another OpenOCD process is not running. Check that you have the correct target and JTAG device plugged in. ****
Resetting target
make[1]: *** [run] Error 1
    make: *** [main_app] Error 2

I have OpenOCD installed

emilf, 2015-10-12:
Try running this and see if it gives you a meaningful error message:
`"./tools/OpenOCD/OSX/openocd-all-brcm-libftdi" -f ./tools/OpenOCD/stlink-v2.cfg -f ./tools/OpenOCD/stm32f4x.cfg -f ./tools/OpenOCD/stm32f4x_gdb_jtag.cfg`

This is assuming you are using stlink-v2 to flash. Also, give the whole command line you are using to build. Try adding VERBOSE=1 to the make command line and add the output of that to the issue and the build/open_ocd.txt file as well. Also, I recommend looking at the WICED forums on Broadcoms site. The issue is more likely to be generic to WICED than specific to WICED-for-EMW and answers to non platform specific issues are often found there.

eliabieri, 2015-10-13:
For the following Command
```
"./tools/OpenOCD/OSX/openocd-all-brcm-libftdi" -f ./tools/OpenOCD/stlink-v2.cfg -f ./tools/OpenOCD/stm32f4x.cfg -f ./tools/OpenOCD/stm32f4x_gdb_jtag.cfg
```
I get this Error. The stink-v2 is attached and connected to a EMW3165
```
Open On-Chip Debugger 0.9.0-00029-g812f48a-dirty (2015-05-28-16:33)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
Info : auto-selecting first available session transport "hla_swd". To override use 'transport select <transport>'.
adapter speed: 1000 kHz
adapter_nsrst_delay: 100
Info : The selected transport took over low-level target control. The results might differ compared to plain JTAG/SWD
adapter_nsrst_delay: 100
Info : Unable to match requested speed 1000 kHz, using 950 kHz
Info : Unable to match requested speed 1000 kHz, using 950 kHz
Info : clock speed 950 kHz
Info : STLINK v2 JTAG v17 API v2 SWIM v4 VID 0x0483 PID 0x3748
Info : using stlink api v2
Info : Target voltage: 3.223857
Error: init mode failed (unable to connect to the target)
in procedure 'init' called at file "./tools/OpenOCD/stm32f4x_gdb_jtag.cfg", line 6
in procedure 'ocd_bouncer'
```
I use the following command to build and flash
```
./make EMW3165-FreeRTOS-LwIP-snip.apsta download run JTAG=stlink-v2
```
And this is the Error
```
Downloading Bootloader ...
**** OpenOCD failed - ensure you have installed the driver from the drivers directory, and that the debugger is not running **** In Linux this may be due to USB access permissions. In a virtual machine it may be due to USB passthrough settings. Check in the task list that another OpenOCD process is not running. Check that you have the correct target and JTAG device plugged in. ****
Downloading DCT ...
**** OpenOCD failed - ensure you have installed the driver from the drivers directory, and that the debugger is not running **** In Linux this may be due to USB access permissions. In a virtual machine it may be due to USB passthrough settings. Check in the task list that another OpenOCD process is not running. Check that you have the correct target and JTAG device plugged in. ****
Downloading Application ...
**** OpenOCD failed - ensure you have installed the driver from the drivers directory, and that the debugger is not running **** In Linux this may be due to USB access permissions. In a virtual machine it may be due to USB passthrough settings. Check in the task list that another OpenOCD process is not running. Check that you have the correct target and JTAG device plugged in. ****
Resetting target
make[1]: *** [run] Error 1
make: *** [main_app] Error 2
```
If I add VERBOSE=1, then this is the error
```
"./tools/common/OSX/mkdir" -p  build/eclipse_debug/
"./tools/common/OSX/echo" Downloading Bootloader ...
"./tools/common/OSX/cp" -f build/waf_bootloader-NoOS-NoNS-EMW3165-SDIO/binary/waf_bootloader-NoOS-NoNS-EMW3165-SDIO.elf build/eclipse_debug/last_bootloader.elf
Downloading Bootloader ...
"./tools/OpenOCD/OSX/openocd-all-brcm-libftdi" -f ./tools/OpenOCD/stlink-v2.cfg -f ./tools/OpenOCD/stm32f4x.cfg -f ./tools/OpenOCD/stm32f4x-flash-app.cfg -c "verify_image_checksum build/waf_bootloader-NoOS-NoNS-EMW3165-SDIO/binary/waf_bootloader-NoOS-NoNS-EMW3165-SDIO.stripped.elf" -c shutdown >> build/openocd_log.txt 2>&1 && "./tools/common/OSX/echo" No changes detected && "./tools/common/OSX/echo" || "./tools/OpenOCD/OSX/openocd-all-brcm-libftdi" -f ./tools/OpenOCD/stlink-v2.cfg -f ./tools/OpenOCD/stm32f4x.cfg -f ./tools/OpenOCD/stm32f4x-flash-app.cfg -c "flash write_image erase build/waf_bootloader-NoOS-NoNS-EMW3165-SDIO/binary/waf_bootloader-NoOS-NoNS-EMW3165-SDIO.stripped.elf" -c shutdown >> build/openocd_log.txt 2>&1 && "./tools/common/OSX/echo" Download complete && "./tools/common/OSX/echo" || "./tools/common/OSX/echo" "**** OpenOCD failed - ensure you have installed the driver from the drivers directory, and that the debugger is not running **** In Linux this may be due to USB access permissions. In a virtual machine it may be due to USB passthrough settings. Check in the task list that another OpenOCD process is not running. Check that you have the correct target and JTAG device plugged in. ****"
"./tools/common/OSX/mkdir" -p  build/eclipse_debug/
"./tools/common/OSX/cp" -f build/EMW3165-FreeRTOS-LwIP-snip_apsta/binary/EMW3165-FreeRTOS-LwIP-snip_apsta.elf build/eclipse_debug/last_built.elf
**** OpenOCD failed - ensure you have installed the driver from the drivers directory, and that the debugger is not running **** In Linux this may be due to USB access permissions. In a virtual machine it may be due to USB passthrough settings. Check in the task list that another OpenOCD process is not running. Check that you have the correct target and JTAG device plugged in. ****
"./tools/common/OSX/echo" Downloading DCT ...
Downloading DCT ...
"./tools/OpenOCD/OSX/openocd-all-brcm-libftdi" -f ./tools/OpenOCD/stlink-v2.cfg -f ./tools/OpenOCD/stm32f4x.cfg -f ./tools/OpenOCD/stm32f4x-flash-app.cfg -c "verify_image_checksum build/EMW3165-FreeRTOS-LwIP-snip_apsta/DCT.stripped.elf" -c shutdown >> build/openocd_log.txt 2>&1 && "./tools/common/OSX/echo" No changes detected && "./tools/common/OSX/echo" || "./tools/OpenOCD/OSX/openocd-all-brcm-libftdi" -f ./tools/OpenOCD/stlink-v2.cfg -f ./tools/OpenOCD/stm32f4x.cfg -f ./tools/OpenOCD/stm32f4x-flash-app.cfg -c "flash write_image erase build/EMW3165-FreeRTOS-LwIP-snip_apsta/DCT.stripped.elf" -c shutdown >> build/openocd_log.txt 2>&1 && "./tools/common [... cut ...]
Info : The selected transport took over low-level target control. The results might differ compared to plain JTAG/SWD
adapter_nsrst_delay: 100
Info : Unable to match requested speed 1000 kHz, using 950 kHz
Info : Unable to match requested speed 1000 kHz, using 950 kHz
Info : clock speed 950 kHz
Info : STLINK v2 JTAG v17 API v2 SWIM v4 VID 0x0483 PID 0x3748
Info : using stlink api v2
Info : Target voltage: 3.223857
Error: init mode failed (unable to connect to the target)
in procedure 'init'
in procedure 'ocd_bouncer'
```

emilf, 2015-10-13:
Y'know, I think your problem isn't OSX. You are talking to the stlink adapter, but it's not talking to your EMW module. Verify your connection from the stlink-v2 to the EMW module. You need to have the EMW module powered and the stlink connected to swdio, swclk, gnd and reset.
Also, look at the serial console, where you can see if the unit is booting with the original factory firmware.
There are photos and a drawing of the connections here: http://www.seeedstudio.com/recipe/344-programming-emw3165-with-broadcom-wiced-and-gcc.html

emilf, 2015-10-22:
9 days without comment. I'm assuming I was right and that this is solved.

closed 2015-10-22 by emilf

eliabieri, 2015-10-23:
I'm sorry I didn't reply.
It all works now, thank you!

-- 

Elia Bieri
whateveryouwant@eliabieri.com

emilf, 2015-10-23:
Cool, thanks for letting me know