MXCHIP-EMW/WICED-for-EMW #15: Using LwIP can make TCP connections and/or re-transmits fail. (upstream issue

emilf, 2015-11-03:
This is an upstream WICED issue,  but serious enough for me to want to address it here.

Reproduction steps:
1. Edit `apps/snip/tcp_client` and change the `include/default_wifi_config_dct.h` access point setting to fit your environment.
1. Run `./make EMW3162-FreeRTOS-LwIP-snip.tcp_client download run`
1. Watch the serial console output.
1. Try starting the echo server by running `python27 apps/snip/tcp_client/tcp_echo_server.py`
1. Notice how the echoing doesn't start working.
1. Reset the EMW module
1. Notice the echo working
1. You can at this point try killing the echo server and restarting it and see how it fails to start echoing again, but
1. Run `./make EMW3162-snip.tcp_client download run` to build ThreadX-NetX instead of FreeRTOS-Lwip and continue from step 2. See how the echoing resumes when the server becomes available.

I reproduced this on EMW3162, but as this seems to be an upstream issue, reported on the WICED forums, this most likely applies to the EMW3165 too.

emilf, 2015-11-04:
I started a thread on the WICED forums about this: https://community.broadcom.com/message/20962#20962

emilf, 2015-11-24:
Broadcom reached out privately and asked if I could reproduce this on WICED 3.4, so this is pending porting to 3.4

ammaree, 2015-11-24:
I suggest they seriously consider to upgrade both FreeRTOS and LwIP to latest versions as part of 3.4 it might solve a number of their problems.

As far as the private extension done by Broadcom to FreeRTOS, they can use what I have incorporated into 8.2.2 (and uploaded to their forum) or even better would be to engage Richard Barry at Real Time Engineers to consider incorporating their changes into the next official release. I have already raised the topic with him.

Andre

emilf, 2015-11-24:
If you drop some links to the relevant discussions, I'll raise that with the guy that contacted me from Broadcom